debug-edge/DebugEdge #1: Top silkscreen on host connector

k0d, 2020-08-29:
It feels like it's too much silkscreen, takes your eye away from the rest of the board.

Here is my suggestion.

[image]

nitz, 2020-08-29:
I'll let @tannewt chime in if he'd like, but in my opinion I feel like that's a bit of the point: to draw your eyes to it and be immediately recognizable.

I also think your render in particular makes it a bit more difficult to see there, since your pads are colored the same as the silk. When the pads are actually gold or silver, I think they stand out a bit more: 
[image]

k0d, 2020-08-29:
[image]

This is what I think would look better. It looks great on the board I'm making...

k0d, 2020-08-29:
Maybe the targets silkscreen isn't great on my above render...

I also wonder...maybe we'd like to include some text or a little 'logo'? then it's even more recognisable....

nitz, 2020-08-29:
Yeah I did a very quick mspaint mockup where I just plopped a random little gear/arrow logo on one of Scott's images: 
[image]

I'm definitely no artist, but I love the idea of a tiny logo that is an immediate "oh i know what this is" hint.

k0d, 2020-08-29:
Leave that to me...I'll produce some mock-ups in a couple of hours. :)

k0d, 2020-08-29:
Suggestion for the host connector.

[image]

[image]

nitz, 2020-08-29:
Ooh, that's fun! I definitely like the spider too. In my quick hunt for "debugging" related iconography I kept coming across things like this that I found on nounproject: [image])

Somehow the spider comes off a lot less "creepy" than the fewer-legged one. Plus I hear they're pretty good at eating bugs!

k0d, 2020-08-29:
haha...yeah, also...the target connector could have a web around it or something...not started to design that yet.

@tannewt your opinion?

k0d, 2020-08-29:
Host and target

Probably the arrows on the 'web' I'll change them...

[image]

k0d, 2020-08-29:
[image]

tannewt, 2020-08-29:
These look like neat ideas. The spider is a great idea! We could use that as the general logo too.

My main goal with the heavy silkscreen was to make the top easily distinguishable from the bottom.

I don't think we'll know what works best until we start seeing it on real boards. All of these renders are much larger that it will be in real life.

nitz, 2020-08-29:
That's a really good point. I swear I need to get a to-scale model of a quarter and keep it every board I do. Sounds like the first spins at oshpark are already at the fab so hopefully we'll get it a good look quickly!

k0d, 2020-08-29:
[image]

This is a minimalist version of it.

I'm going to order a prototype board for a project tomorrow, I'll order a board with each of my examples to compare.

Yes, I was thinking we can use the spider as a logo also.

tannewt, 2020-08-29:
@k0d do you not like the slanted pads? I like having a version that would work without silkscreen.

k0d, 2020-08-29:
[image]

Personally...no I don't like them...but I agree with you...

k0d, 2020-08-29:
It's hard to get the spider+text to be nice with the slants...but I'm working on it...

tannewt, 2020-08-29:
I'm ok omitting the text too and just having the spider.

nitz, 2020-08-29:
I agree with that. The spider is something that would be recognizable, and something someone could google "spider connector" or something like that 😂

k0d, 2020-08-29:
[image]

What do you think of the bottom left one? I will fix the slanted pads...

k0d, 2020-08-29:
[image]

Green + HASL. (This is what my next board will be...so I am keen to make it look nice on it)

k0d, 2020-08-29:
[image]

k0d, 2020-08-29:
[image]

I feel this is the 'final design' from my point of view...so now it's up to you to let me know what you think, should I change anything and if you want to use it.

nitz, 2020-08-29:
I actually _really_ like it. The IC shape on the target side side is super cute. I don't know how it would work at that scale, but you gave me the idea of trying to put like, a "webbing" inside of it. Excusing my (very) poor mspaint rendition:

[image]
(Or like you have with the chip, then use mask to 'paint' the web lines?)

Again, I know it's super tiny and I'm not sure how well it'd look at that scale, but I feel like it would hit the theme spot on.

You also gave me an idea for a swankied up wordmark with the way you did the in/out style like that:

[image]

k0d, 2020-08-29:
I think its a no with the web....too small, am already pushing the limits as it is.

The Logo...nice start. Will think about that tomorrow...

tannewt, 2020-08-30:
I really like the spider on one side and the ic on the other!

k0d, 2020-08-30:
[image]

I've changed the silkscreen a little...and optimised the images...as it took like 1 minute to do the copper flood fill command as there was so many edges for it to work around...now it's normal speed.

If you like this...I'm going to order a prototype of a board I'm going to be selling...which I want to use this connector on. I understand it's still work in progress, but I'd like to get as close to a real connector as possible so any prototype users will not get confused if it's changed a lot.

tannewt, 2020-08-30:
> Screenshot 2020-08-30 at 20 08 08 <https://user-images.githubusercontent.com/547698/91666384-8909cb00-eafc-11ea-87ab-6494b866a4c7.png>

I was thinking the slant would peak at the slot. For the target side I wanted a slant as well as to make the gnd pin stick out further to “fit” into the host “plug”.

k0d, 2020-08-30:
Ok...I need a bit more info about that, I don't really understand what you want to achieve with it...

tannewt, 2020-08-30:
Sorry. I’m on my phone so I’m a bit limited. For the target, I was going to set back all pins except gnd by the same amount as the gnd pin on the host. That help? Otherwise I can try and implement it later.

k0d, 2020-08-30:
Ok, I get that...was there a technical reason, or purely cosmetic?

tannewt, 2020-08-30:
Mostly cosmetic. It will also connect the gnd pin before the others when the target is connected second which is better I think.

k0d, 2020-08-30:
[image]

Is this ok for host? I did the slant going in because otherwise the spider would stick too far into the rest of the PCB and take up unnecessary space.

tannewt, 2020-08-30:
> Screenshot 2020-08-30 at 20 55 40 <https://user-images.githubusercontent.com/547698/91667226-2d8f0b80-eb03-11ea-82b5-775f25efa322.png>

I like it!

nitz, 2020-08-30:
Same! I'm good with taking up less space that way for sure!

I can't get over how much I dig that little spider!

k0d, 2020-08-30:
If we want to connect GND first, shouldn't it be longer on the host side also?

tannewt, 2020-08-30:
Yes but I’m more worried about it being obvious which is host and which is target.

k0d, 2020-08-30:
Host = Debugger = Spider with his arrow pointing to the ....
....arrow which points to ... Target = MCU

k0d, 2020-08-30:
[image]
 ok ?

tannewt, 2020-08-30:
I understand that. I worry that detail is too small and that the of differences would be more immediately obvious.

tannewt, 2020-08-30:
> Screenshot 2020-08-30 at 22 21 22 <https://user-images.githubusercontent.com/547698/91668734-266dfa80-eb0f-11ea-9307-2950c20c0ecd.png>

The host side should be indented while the target would have a longer gnd. I liked the host before.

k0d, 2020-08-30:
A lot of target boards would use 6 pin I think...at least mine would...SWD and SWO are probably most common...unless it's full jtag like esp32 etc...but still...I'll make the target more different also.

tannewt, 2020-08-30:
Right. I’m thinking the 10 pin design will apply to the fewer pin designs as well. The spider and the gnd pin are always present.

k0d, 2020-08-30:
[image]

I had this idea also...and invert the arrow on the target....if you don't like it, I'll go back to the short host GND pad...

tannewt, 2020-08-30:
> Screenshot 2020-08-30 at 22 29 38 <https://user-images.githubusercontent.com/547698/91668853-4ce06580-eb10-11ea-8666-bb3f9d10cf72.png>

Ya I like that! Good idea

k0d, 2020-08-30:
[image]

It didn't look nice having the arrow point 'in' on the target...what about this?

k0d, 2020-08-30:
[image]

Clearance of the connector...you still can see the 'icons'...

k0d, 2020-08-30:
[image]

[image]

[image]

"Final version" ;)

tannewt, 2020-08-30:
> Screenshot 2020-08-30 at 23 36 28 <https://user-images.githubusercontent.com/547698/91669983-a26d4000-eb19-11ea-8448-2e393e0a9df0.png>
> Screenshot 2020-08-30 at 23 38 48 <https://user-images.githubusercontent.com/547698/91670024-f6782480-eb19-11ea-9b28-612154a00428.png>
> Screenshot 2020-08-30 at 23 40 01 <https://user-images.githubusercontent.com/547698/91670052-232c3c00-eb1a-11ea-989c-e9cba799d5b1.png>
> "Final version" ;)

Looks great!

nitz, 2020-08-31:
Absolutely fantastic. I really appreciate all the work you're putting into this. The footprint looks so dang awesome!

tannewt, 2020-08-31:
@k0d You should have permissions to push all of this to the repo.

k0d, 2020-08-31:
Yeah....will tidy things up, and push in the next couple of days.

k0d, 2020-09-12:
I have boards now, they work well. However the silkscreen and spidey/mcu logos are a bit hard to see. I want to make some small changes, but nothing major.

k0d, 2020-09-12:
Quite a big change...but it'll be much easier to recognise the footprint.

[image]

k0d, 2020-09-12:
OLD:
[image]
NEW:
[image]

k0d, 2020-09-12:
I like this....and you @nitz @tannewt ?

[image]
[image]

tannewt, 2020-09-12:
> I like this....and you @nitz <https://github.com/nitz> @tannewt <https://github.com/tannewt> ?
> Screenshot 2020-09-12 at 09 36 41 <https://user-images.githubusercontent.com/547698/92990296-85eae380-f4db-11ea-89fc-8bbe0f421b12.png>
> Screenshot 2020-09-12 at 09 35 08 <https://user-images.githubusercontent.com/547698/92990293-82575c80-f4db-11ea-8de3-7e72787f4c6e.png>

I like that the arrow is clearer but why straighten the pads?

k0d, 2020-09-12:
I had to make the MCU bigger...so the pads had to be straight...but on the back...they're still pointed.

I think this makes it clearer which side is which.

tannewt, 2020-09-12:
Ok! Thanks for iterating on the design.

k0d, 2020-09-12:
[image]
[image]

Ok, so I think I'm done with my updates. I like the MCU a lot more. It will be way easier to get fab'd, I've also updated Spidey a little so he's more fab'able.

nitz, 2020-09-12:
Is the center of the IC art just bare FR4? Cause that's super clever!

I kinda liked the way the spider was "holding" the arrow" before, but I will definitely yield that it is certainly clearer.

I'm thrilled with your artistic ability :)

[1 more reply]

closed 2020-09-28 by k0d