debug-edge/DebugEdge #2: Pin 1 silkscreen

k0d, 2020-08-29:
The purpose of the marking for pin 1 in the case of this connector is to show if it's up or down. But the connector has a logo on the top, and an arrow pointing at their pin 1. So I suggest we change the silkscreen to show what is up...not which is pin 1.

**'The issue'**
[image]

**'The fix'**
[image]

nitz, 2020-08-29:
The pin 1 marking is definitely partially a "this side up" indicator (and is definitely not any more than that on the first oshpark spins.) The shortening/lengthening of pin 1 on the host/device sides respectively was thought to keep that "this side up", as well as ideally be a "ground is connected before everything else" feature, assuming you're attaching to the target second.

As it is now, the bottom side of the FP has no silk whatsoever, so any silk would signify the top, in addition to the shortened/lengthened pin1.

I wish we could stay 'in sync' with the AVX pin one, but their pin 1 varies based on the number of positions in the connector, and we want to "grow" from the 4p to the 10p, only adding pin positions rather than moving/adding.

k0d, 2020-08-29:
Yeah...I totally agree that it'd have been nicer to match their pin 1, but as we can't...I guess it's good to just forget indicating pin 1.

In general, I'm a "minimalist silkscreen guy", so you'll see why I wanted to make it a bit less...

k0d, 2020-08-29:
Closing because #1 can cover both issues.

closed 2020-08-29 by k0d