debug-edge/DebugEdge #3: Orientation of connector

k0d, 2020-08-29:
As you can see in the photo below, the logo on the connector is closest to the target side, not the host side. Is that what we want? It's ok if it is :) Just making sure.

[image]

nitz, 2020-08-29:
It honestly wasn't really a consideration when I started laying out the footprint. I figured it had to go on one of the sides, and I typically have my debugger/flasher sitting "above" my board on my desk, and had just imagined it all from that angle, if you will.

I'm not sure if there's really a right answer though. My biggest concern is just making sure it's connected correctly, I think.

k0d, 2020-08-29:
I agree with you....just wanted to check that it was considered and you're ok with it. Not sure if you tested with a 3d model...

nitz, 2020-08-29:
Haha, nope, I hadn't. I have a little 3D printed origin axis model I keep on my desk and used it to try and visualize the rotations, but that was about as far as I got 🙂

Though I'm gonna keep this open now because it makes sense to add one to the kicad preview.

k0d, 2020-08-29:
The thing with the preview is you want to see it with and without the connector...that's why I did it half on...but I'm not sure...I know you can turn off the render of the 3d models...not sure what's best...I don't have much of an opinion on that.

siddacious, 2020-08-31:
This caught my attention because it reminds me of how trivial but also satisfying it is to have finally figured out the "best" way to plug in USB type A connectors, with the thick/male side "down". I'm also noticing just now that the logo on type A cables is seemingly always "up" or on the side opposite the contacts/male part.

With that in mind, to the extent that there is an "up" or side that should be a focus, _ideally_ the logo would be on the same side. Other considerations should likely take precedence, but it's worth considering, I think.

k0d, 2020-09-12:
You don't need to look at the logo/arrow on the connector as you can put it either way round...you orientate it according to the footprint design.

closed 2020-09-12 by k0d

siddacious, 2020-09-15:
oh derp, ya there's only one sensible way to plug 'er in. I have eyes, I swear!