debug-edge/DebugEdge #4: VRef on the connector

k0d, 2020-08-31:
VRef is used on the JTAG connector to detect the target's voltage. It shouldn't be used to supply power to the target.

So, if we want to provide power with debug-edge, then we need to change it to VCC. Or if we use it as VRef, then the connector is still BYOP.

I prefer to have VRef, so it's then more compatible with the ARM/JTAG 10 pin connectors.

Maybe confusing and not 100% checked yet...but here is the pinouts of various connectors.

[image]

nitz, 2020-08-31:
Yeah, Vref may be a bit of a misnomer, but I'll explain how we got there and see what you think:

I started from the JTAG/SWD interface [directly from Keil's ULINKplus](https://www.keil.com/support/man/docs/ulinkplus/ulinkplus_jtagswd_interface.htm) In that, they call it VCC. The CMSIS-DAP (DAPLink) adapter I was currently using (as in, sitting on my desk at the moment) also uses VCC as one of it's 5 pins (VCC,GND,SWDIO,SWCLK,nRESET).

I think it was when @tannewt I were looking at seeing that it could support lots of different protocols that we swapped it to Vref, because it didn't _necessarily_ need to supply power in some of those cases. I really can't tell if it's something we should be naming "VCC/Vref", naming something else, or how to deal with it exactly.

If you want a real confusing rabbit hole to dig down, check out all the different names/'functions' [ST](https://www.st.com/resource/en/user_manual/dm00026748-st-link-v2-in-circuit-debugger-programmer-for-stm8-and-stm32-stmicroelectronics.pdf) uses for essentially the same pin: VDD, Target VCC, MCU VCC, MCU VDD, VAPP.

Maybe we need a whole new name that is all of those, *plus* the potential of Vref?

k0d, 2020-08-31:
On the STLink v3 board, it's connected to a voltage divider that the STLink MCU uses to detect the target voltage.

If we called it VCC, then I'd expect it to supply voltage...which a STDC14 to Debug-Edge adapter board would not. On JTAG there is a pin (19 I think) which is 5v supply, that's VCC in my view.

What's the CMSIS-DAP connector you have?

This one https://github.com/ARMmbed/mbed-HDK-Eagle-Projects/blob/master/DAPLink/Probes/SWDAP/v1.0.0/SWDAP-v1.0.0-SCH.pdf is kind of an official one, and VCC isn't connected. It's meant to be T_VCC...but they don't label it as that it seems.

OpenOCD uses the target voltage detection sometimes to do then flash programming. It can fail sometimes if you don't use it.

My suggestion is we keep VRef and use it as T_VCC (don't connect to H_VCC), but maybe it's good to get other people into this discussion. Maybe @ladyada @arturo182 ?

k0d, 2020-08-31:
Hmm, off course, if you supply 3v3 on the VRef to the target, then the target's VCC will be 3v3, which is what it'd detect. So I guess it depends on if the H_VCC line can supply enough mAh to the target.

I think it's ok to use it as both....but...we should clearly mark that VRef might not supply current in the debug-edge docs. That's the biggest issue with other connector specs, it's not clearly communicated.

k0d, 2020-08-31:
Oh....should we add a marking on the VRef pin to indicate if it supplies power? then the board developer can choose to show it or not.

nitz, 2020-08-31:
^ That was along the lines of what I was thinking, but I couldn't come up with a clean (read: tiny) way to say "this *might* supply power, depending on your adapter/protocol/etc." Any ideas?

k0d, 2020-08-31:
My board I'm making right now will not supply power, so it doesn't matter for me today.

nitz, 2020-08-31:
That's good at least! Yeah that's why I had the little solder bridge there
that can be cut on the Vref line on the host board side, because I'm not
quite sure what the right answer is yet, too!

tannewt, 2020-08-31:
I wasn't thinking it'd supply power. There should definitely be a way to power it separately.

closed 2020-09-28 by k0d