debug-edge/DebugEdge #7: Slot issues

k0d, 2020-09-01:
JLCPCB can't produce the 0.6mm slot as their minimum is 1.0mm.

So I don't know if we should widen the slot, or if I should move to another connector on my boards. It's quite important I can use JLCPCB....

k0d, 2020-09-01:
[image]

I think I'm going to try this...

k0d, 2020-09-01:
[image]

Target

nitz, 2020-09-01:
Hm. Let me see if I can find something with a 1mm slot around the office here, and I'll see how much play the adapter has when I get home.

I will say, the grip it has on the board itself (with just the plastic, not to mention the pins), makes me think that it's *probably* not  an issue for a wider slot.

A fun fact, OshPark told me they actually had to modify some of their tooling for the slot too. Seems like the 0.6mm is a bit of a wrench in PCB manufacture's plans! No chance we can get JLCPCB to be as accommodating as Oshpark? ;D

k0d, 2020-09-01:
I ordered the boards...so I'll be able to test properly in a week or so....Hopefully it'll be fine.

tannewt, 2020-09-06:
I think the 1mm slot is fine. @k0d can you push your latest footprints? I want to add it to some boards too.

k0d, 2020-09-06:
Sure, I'll do it today. I was waiting until my boards came, that's why I didn't do it yet ;)

nitz, 2020-09-07:
Looks like my oshpark boards should be in tomorrow, so I'll get a chance to see how the 0.6mm slot works. If I'm feeling squirrely, on Wednesday at the office I might take the dremel to one and widen it out to see how it feels too! (assuming I don't completely botch the pads 😂

k0d, 2020-09-12:
1.0mm slot works just fine.

closed 2020-09-12 by k0d

nitz, 2020-09-12:
Hell yeah! Great stuff, Mark!