debug-edge/DebugEdge #12: jtag/uart pinout suggestion

gsteiert, 2020-09-14:
I like where this is going, but it would be really nice to fit the uart in the 8 pin option. 
I would suggest this as an enhanced pinout for jtag/uart options:
1 GND
2 SWDIO/TMS
3 SWCLK/TCK
4 nRESET
5 Vref
6 SWO/TRACE_CLK
7 TDI/RX
8 TDO/TX
9 TRACE0
A TRACE1

This would allow uart in the 8 pin option and all the pin directions will match. 
Also, most chips only have swo or trace, so sharing the swo pin with trace makes more sense (I know that is not how arm did it)

nitz, 2020-09-15:
This makes pretty decent sense to me. I'm not super familiar with JTAG myself, but it seems plausible that if you're using TDI/TDO you're either already using the trace pins or have other means of getting the UART.

I also appreciate the symmetry of lining up the RX/TDI (both would be data to target from host), and TX/TDO (to host from target).

tannewt, 2020-09-27:
I purposefully avoided sharing JTAG with UART so that one could use them concurrently. My compromise was to have TX on the 8 pin so that debug output could be done at least.