debug-edge/DebugEdge #14: Create a logo for debug-edge

k0d, 2020-09-27:


k0d, 2020-09-27:
[image]

I made one...like it?

k0d, 2020-09-27:
[image]

...or we could go blue?

k0d, 2020-09-27:
[image]

...or even green?

nitz, 2020-09-27:
Ahhh, I like it. I'm always a fan of purple not only because of oshpark, but also cause pink and purple are my favorite colors, but they honestly all look fantastic. 

Also blue should be a more common pcb color, it's so fun.

cotdp, 2020-09-27:
+1 for purple :)

k0d, 2020-09-28:
Thing is, with purple, everyone thinks of OSHPark, which concerns me only because we're not associated with them. I kinda like the blue one. By the way...these are not made in KiCAD, but 'sketch', so they're proper artwork.

[image]

Here's a white version for if we want a dark background. I won't make yellow, red or black.

k0d, 2020-09-28:
[image]

@nitz Here's one just for you

nitz, 2020-09-28:
You spoil me 🥰🥰

k0d, 2020-09-30:
@tannewt your views?

tannewt, 2020-10-05:
Sorry I missed this. Where do you want to use the logo? Seems to me that is should be simpler. Maybe just the spider pointing to the chip?

k0d, 2020-10-05:
I was thinking top of the webpage for example...but I do think having a simpler icon is also good.
So what I've designed so far is a 'titlebar', and I'll make an 'icon'. Sound good?

tannewt, 2020-10-06:
Sure, I'd start with the logo and then create the rest of the design.