debug-edge/DebugEdge #24: DebugEdge_2x03_Target.kicad_mod fails design rules checker in Kicad 8.0

jcard0na, 2024-12-25:
I just found this very cool project.  Thanks for making it available to the world!

When I tried to use it in a new design, I get a bunch of Design Rules Checker error in Kicad that are triggered by the cute logo:

[image]
[image]

The fix is simple:  just remove or move to silkscreen like below.

[image]

If anyone is maintaining this, I'm happy to provide a pull request.

Cheers,

tannewt, 2024-12-26:
I'm around to merge. Haven't been using it myself though.

jcard0na, 2024-12-26:
Thanks @tannewt 
If I end up building a board with this, I'll send a PR.  For the moment, at least this issue documents the problem and provides one possible solution.